adafruit/Adafruit-Raspberry-Pi-Python-Code #106: Using i2c library

Marsfan, 2015-01-04:
Ok, before I start this, let my just say that I am a total NOOB, I just got my my for Christmas, and I am still learning how to use it (but I do know how to use Arduino). 

How do I use the adafruit i2c Python library? there seems to be no documentation on it.

alexellis, 2015-01-04:
What sensor are you wanting to work it? There is usually an example of using the library - or are you hoping to work with a sensor that has no example yet?

Marsfan, 2015-01-05:
Ok sorry, I do not have a particular sensor, I just would like to know the different functions/syntax/API for using it. I'm hoping to use i2c to talk to a variety of different things, but my main plan is to use an arduino as an analog to digital converter/port expander.

b1weber, 2015-01-11:
Hi Marsfan, from one noob to another, these libraries are really valuable!
Here is how I use it to read the pressure and pressure every 5 minutes from my pi and a BMP085;
``` 
from Adafruit_BMP085 import BMP085
import MySQLdb
db=MySQLdb.connect(host="x.x.x.x",user="",passwd="",db="dbname",port=int(3306))
sql="Insert Into Sensors(temperature, pressure) Values(%s, %s);"
bmp=BMP085(0x77)
temp=bmp.readTemperature()
tempf=(9.0 / 5.0) * temp + 32
pressure=bmp.readPressure()
cursor=db.cursor()
cursor.execute(sql, (tempf, pressure))
db.commit()
```
The I2C is handled within the sensor libraries.
This is the sensor I am using
https://www.adafruit.com/product/391
I hope this help!
Bob

Marsfan, 2015-01-11:
Thanks for posting the reply, but I am looking for a guide on how to handle just the  I2C part, I do not want to use any of the sensors libraries. 
From what I can see, this does come with an I2C library, that's what I would like to know how to use.

pabloocaldeira, 2018-06-20:
Someone can help me use Adafruit_MCP230xx.py library with 2 or more MCP23017 (I/O expander)? I tried some ways but without succeed, only with a single MCP23017, according to the tutorial: https://cdn-learn.adafruit.com/downloads/pdf/mcp230xx-gpio-expander-on-the-raspberry-pi.pdf

Marsfan, 2018-06-21:
@pabloocaldeira Probably better to start a new thread. But have you set the MCP23017 chips to have different addresses?

Marsfan, 2018-06-21:
Also, I realized that I never closed this. I figured out what I was working on three years ago. Closing now.

closed 2018-06-21 by Marsfan