adafruit/Adafruit-Raspberry-Pi-Python-Code #114: Adafruit_PWM_Servo_Driver broken; "IOError: 13" on importing module

Symmetric, 2015-03-30:
The PWM servo driver module is broken. It's producing this error when I import it:

```
Traceback (most recent call last):
  File "/home/pi/git/BodySensor/PiDispatcher/venv/bin/dispatcher_server", line 9, in <module>
    load_entry_point('dispatcher==0.0.1', 'console_scripts', 'dispatcher_server')()
  File "/home/pi/git/BodySensor/PiDispatcher/venv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 519, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/pi/git/BodySensor/PiDispatcher/venv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2630, in load_entry_point
    return ep.load()
  File "/home/pi/git/BodySensor/PiDispatcher/venv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2310, in load
    return self.resolve()
  File "/home/pi/git/BodySensor/PiDispatcher/venv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2316, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/pi/git/BodySensor/PiDispatcher/dispatcher/server.py", line 5, in <module>
    from dispatcher.Adafruit_PWM_Servo_Driver import PWM
  File "/home/pi/git/BodySensor/PiDispatcher/dispatcher/Adafruit_PWM_Servo_Driver.py", line 11, in <module>
    class PWM :
  File "/home/pi/git/BodySensor/PiDispatcher/dispatcher/Adafruit_PWM_Servo_Driver.py", line 35, in PWM
    general_call_i2c = Adafruit_I2C(0x00)
  File "/home/pi/git/BodySensor/PiDispatcher/dispatcher/Adafruit_I2C.py", line 43, in __init__
    self.bus = smbus.SMBus(busnum if busnum >= 0 else Adafruit_I2C.getPiI2CBusNumber())
  File "/home/pi/git/BodySensor/PiDispatcher/venv/local/lib/python2.7/site-packages/smbus/smbus.py", line 119, in __init__
    self.open(bus)
  File "/home/pi/git/BodySensor/PiDispatcher/venv/local/lib/python2.7/site-packages/smbus/smbus.py", line 146, in open
    raise IOError(e.errno)
IOError: 13
```

Looking at the code, it's assuming that there is an i2c register at 0x00 that it can create a PWM object for. That's not the case on my system (using the latest Raspbian Wheezy circa this post, and the Adafruit 16-channel PWM microcontroller).

My valid I2C addresses are:

```
$ sudo i2cdetect -y 0                                                                                                                                                                      
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
```

Is there some requirement that I'm missing to get this working? Or is the code making an assumption that doesn't hold on all platforms?

tdicola, 2015-03-30:
Can you run these commands to check what I2C bus is available, and what revision is your Pi?

```
ls -l /dev/i2c*
```

Then run this to get the Pi revision and other info:
```
cat /proc/cpuinfo
```

If you can paste the output of both back in the thread here we can troubleshoot a little more.  Normally the library reads the revision info from /proc/cpuinfo to figure out if it should use I2C bus 0 or 1.  Very old Pi's used I2C bus 0 (like what you see with your i2cdetect command) and newer ones use bus 1.  It looks like something might be getting confused with the revision detection logic and using the wrong bus.  Let's check the output of those commands to be sure though, thanks!

Symmetric, 2015-03-31:
Turns out this is just what happens when you don't have permission to access the i2c device.

closed 2015-03-31 by Symmetric