adafruit/Adafruit-Raspberry-Pi-Python-Code #132: python3 compatibility

Talv, 2015-10-16:
Any plans for supporting python3?

Also, there's some inconsistency about the indention in *Adafruit_ADS1x15.py* (and not only). Spaces are mixed with tabs. Even incorectly, as there's one tab for 4 spaces, while the file indention is 2 spaces. As seen on the picture:
[image]
That's perhaps topic for another issue but it bothers me much while converting it to py3.

timluther, 2015-10-24:
Seconded, it's easy to write python3 style code that's still compatible with python2.x. Can developers please add parentheses to their print statements and kill the commas on their exceptions?

scls19fr, 2016-01-24:
+1 a simple search in Github https://github.com/adafruit/Adafruit-Raspberry-Pi-Python-Code/search?utf8=%E2%9C%93&q=print helps to find these kind of python 3 incompatibilities

An other way to do it is to install `flake8-print` https://github.com/jbkahn/flake8-print

```
here is an example:

print "statement"

print("function")
```

With Python 2
```
$ flake8 main.py
main.py:1:1: T001 print statement found.
main.py:3:1: T001 print function found.
```

With Python 3

```
$ flake8 main.py
main.py:1:18: E901 SyntaxError: invalid syntax
```

ladyada, 2016-01-24:
@tdicola has a Plan to do a major cleanup of this creaky repo including pep8, py3, and pypi...its quite a pppparty :)

closed 2016-12-28 by Talv