adafruit/Adafruit-Raspberry-Pi-Python-Code #144: A package for Adafruit-Raspberry-Pi-Python-Code

scls19fr, 2016-01-23:
Hello,

it will be nice to have package for Adafruit-Raspberry-Pi-Python-Code.

Kind regards

Liambeguin, 2016-01-24:
Hi, I don't know if it's the exact same code but a package is available.
```
sudo pip install Adafruit_Libraries
```
https://pypi.python.org/pypi/Adafruit_Libraries/0.1

mecworks, 2016-01-24:
It's definitely not the same code as it was last updated in 2013.  It would
be nice if there were a cronjob that kept the pip version up to date.

scls19fr, 2016-01-24:
A `__init__.py` in each directory is necessary. A `setup.py` file is also required.

Liambeguin, 2016-01-24:
the ```setup.py``` file can be taken from the current package and updated

scls19fr, 2016-01-24:
`setuptools` might be prefered instead of `distutils`

scls19fr, 2016-01-24:
A LICENCE.txt file is necessary.
I also noticed that README is a Markdown file (but PyPi doesn't render them correctly...). Maybe using RST might be considered.

I also noticed that in this package most directory was renamed

```python
from setuptools import setup, find_packages
NAME = 'Adafruit_Libraries'

setup(
	name=NAME,
	version='0.x.y',
	author='Adafruit',
	packages=find_packages(),
	url='http://www.adafruit.com/',
	license='LICENSE.txt',
	description='Libraries for using Adafruit Hardware on a Raspberry Pi or BeagleBone Black',
	long_description=open('README.md').read(),
)
```

scls19fr, 2016-01-24:
I've just send a PR which provide a package. Pinging @ladyada @tdicola 
Could you have a look ?

ladyada, 2016-01-24:
@tdicola has a Plan to do a major cleanup of this creaky repo including pep8, py3, and pypi...its quite a pppparty :)

mecworks, 2016-02-09:
I was wondering how this is coming.  I checked the current project and no __init__.py files exist yet so I assume the package-ability is not there yet.  Will these files be in the git repo or only in the package on pypi/pip?  Also, is there an ETA for a pypi/pip package?

ladyada, 2016-02-09:
https://github.com/adafruit/Adafruit_Python_ADS1x15
and
https://github.com/adafruit/Adafruit_Python_MCP3008
are done

mecworks, 2016-02-10:
So, each module will be its own package rather than one package that contains all Adafuit modules?  I'm fine either way.  Separate packages are easy to install just what's needed for a project, a single package that contains all of Adafruit's packages would ensure you have all packages and that they all work with each other.

ladyada, 2016-02-10:
yep each package will be split off, it will make maintenance and cross-dependancies way easier.
there is no ETA when they'll be done, its a lot of work, we're also fixing bugs and pulls at the same time

mecworks, 2016-02-10:
Very cool.  Thanks for the great work.

scanner, 2016-02-10:
Not sure if this is helpful at all, but I have made two modules based on a combination of the Adafruit arduino code and the Adafruit Python GPIO module for the TSL2561 and HDC100x modules:

scls19fr, 2016-02-17:
If you make one repository per module I suggest to use a template like cookiecutter https://cookiecutter.readthedocs.org/en/latest/

mecworks, 2016-02-17:
Cookiecutter looks pretty cool.  There's already defined cookiecutter
templates for pip projects and python projects.

caternuson, 2016-06-08:
I would've voted for a single package. While it may make maintenance more difficult under the hood, the user experience would be a little cleaner.

A simple example:
```python
from adafruit import GPIO
GPIO.output(23, GPIO.HIGH)
```
Another example:
```python
from adafruit import Matrix8x8
m = Matrix8x8.Matrix8x8()
m.setPixel(4, 2, m.ON)
```

ladyada, 2016-06-08:
@caternuson we tried an it was simply not tractable for our developers. we wanted to keep working on it without going insane :)
also, since we did it and its done, we'll be closing this issue. thx!

closed 2016-06-08 by ladyada