adafruit/Adafruit-Raspberry-Pi-Python-Code #156: Unable to install many of the new python modules using python3

mecworks, 2016-08-08:
I am using Ubuntu 16.04LTS on a desktop computer with both python2.7 and python3.5 installed.  All packages installed correctly using python 2.7 however, when installing using python 3 (pip3), I get the following error for many of the packages as I try to install them.  This one, is the error for the mcp3008 module but all are the same error.

Collecting adafruit-mcp3008
  Downloading Adafruit_MCP3008-1.0.1.tar.gz
    Complete output from command python setup.py egg_info:
    Downloading http://pypi.python.org/packages/source/d/distribute/distribute-0.6.14.tar.gz
    Extracting in /tmp/tmpd5gowb59
    Traceback (most recent call last):
      File "/usr/local/lib/python3.5/dist-packages/ez_setup.py", line 143, in use_setuptools
        raise ImportError
    ImportError
    
    During handling of the above exception, another exception occurred:
  
```python  
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-dr4m3nxh/adafruit-mcp3008/setup.py", line 2, in <module>
        use_setuptools()
      File "/usr/local/lib/python3.5/dist-packages/ez_setup.py", line 145, in use_setuptools
        return _do_download(version, download_base, to_dir, download_delay)
      File "/usr/local/lib/python3.5/dist-packages/ez_setup.py", line 125, in _do_download
        _build_egg(egg, tarball, to_dir)
      File "/usr/local/lib/python3.5/dist-packages/ez_setup.py", line 99, in _build_egg
        _extractall(tar)
      File "/usr/local/lib/python3.5/dist-packages/ez_setup.py", line 467, in _extractall
        self.chown(tarinfo, dirpath)
    TypeError: chown() missing 1 required positional argument: 'numeric_owner'
```

I edited /usr/local/lib/python3.5/dist-packages/ez_setup.py, line 467 and added an empty parameter to satisfy the above error and got further but then ended up with the following:

```python
Collecting adafruit-mcp3008
  Downloading Adafruit_MCP3008-1.0.1.tar.gz
    Complete output from command python setup.py egg_info:
    Downloading http://pypi.python.org/packages/source/d/distribute/distribute-0.6.14.tar.gz
    Extracting in /tmp/tmprw17in6r
    Now working in /tmp/tmprw17in6r/distribute-0.6.14
    Building a Distribute egg in /tmp/pip-build-tz5sq5vx/adafruit-mcp3008
    creating build
    creating build/src
    creating build/src/docs
    creating build/src/docs/_templates
    creating build/src/docs/_theme
    creating build/src/docs/_theme/nature
    creating build/src/docs/_theme/nature/static
    creating build/src/docs/build
    creating build/src/docs/build/html
    creating build/src/docs/build/html/_sources
    creating build/src/docs/build/html/_static
    creating build/src/setuptools
    creating build/src/setuptools/command
    creating build/src/setuptools/tests
    creating build/src/setuptools/tests/indexes
    creating build/src/setuptools/tests/indexes/test_links_priority
    creating build/src/setuptools/tests/indexes/test_links_priority/simple
    creating build/src/setuptools/tests/indexes/test_links_priority/simple/foobar
    creating build/src/tests
    creating build/src/tests/shlib_test
    copying setuptools/__init__.py -> build/src/setuptools
    copying setuptools/sandbox.py -> build/src/setuptools
    copying setuptools/depends.py -> build/src/setuptools
    copying setuptools/extension.py -> build/src/setuptools
    copying setuptools/dist.py -> build/src/setuptools
    copying setuptools/package_index.py -> build/src/setuptools
    copying setuptools/archive_util.py -> build/src/setuptools
    copying setuptools/command/install_scripts.py -> build/src/setuptools/command
    copying setuptools/command/__init__.py -> build/src/setuptools/command
    copying setuptools/command/bdist_wininst.py -> build/src/setuptools/command
[... 245 lines cut ...]
       or: setup.py --help-commands
       or: setup.py cmd --help
    
    error: invalid command 'bdist_egg'
    /tmp/pip-build-tz5sq5vx/adafruit-mcp3008/distribute-0.6.14-py3.5.egg
    Traceback (most recent call last):
      File "/usr/local/lib/python3.5/dist-packages/ez_setup.py", line 143, in use_setuptools
        raise ImportError
    ImportError
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-tz5sq5vx/adafruit-mcp3008/setup.py", line 2, in <module>
        use_setuptools()
      File "/usr/local/lib/python3.5/dist-packages/ez_setup.py", line 145, in use_setuptools
        return _do_download(version, download_base, to_dir, download_delay)
      File "/usr/local/lib/python3.5/dist-packages/ez_setup.py", line 125, in _do_download
        _build_egg(egg, tarball, to_dir)
      File "/usr/local/lib/python3.5/dist-packages/ez_setup.py", line 116, in _build_egg
        raise IOError('Could not build the egg.')
    OSError: Could not build the egg.
```
I am unable to install most if not all of the packages using Python3

mecworks, 2016-09-13:
Has anyone tried to install the new python packages on a raspberry pi for python 3??

I'm still failing using pip3 install adafruit-ads1x15 (or any other of the adafruit libraries)

wolfspyre, 2016-10-14:
I seem to have made progress with this after installing ez_setup inside my venv

````
wget https://bootstrap.pypa.io/ez_setup.py -O - | python && pip3 install Adafruit_gpio
 ...[Snip]...

Successfully downloaded Adafruit-gpio adafruit-pureio spidev
````

tdicola, 2016-10-20:
Apologies it took a while to look at this but it looks like there was an odd interaction with the ez_setup script and Python 3.  Ez_setup is there to install setuptools automatically for python 2, like if you load a Raspberry Pi with Raspian from scratch and don't have pip or anything else installed (unfortunately they don't include the python-setuptools package by default).  The problem is ez_setup doesn't seem to work with Python 3 and ultimately isn't necessary since Python 3 includes setuptools.  Long story short it's a weird quirk of supporting both Python 2 and 3 that can luckily be worked around in the setup.py.

I just updated all the libraries, including this one, on pip to have a fix.  Give a pip3 install of the library a try again (you might want to add the --force option to force it to reinstall in case you installed already).  Thanks for raising the issue and investigating!

closed 2016-10-20 by tdicola